SeongGino/ir-light-gun-plus #2: esp32-c3 super mini

rnrobles, 2023-11-04:
I just bought a couple of these boards and I'm doing some tests like writing eprom

It looks like a good board with good performance and size

Could you help me make it compatible?

esp32-c3 super mini

https://www.aliexpress.com/item/1005005967641936.html?spm=a2g0n.order_list.order_list_main.6.99711802uEbMTj

Core: 32-bit RISC-V single-core microprocessor, frequency up to 160MHz
2.4GHz WiFi compatible with IEEE 802.11 b/g/n
Bluetooth LE: Bluetooth 5, Bluetooth mesh
4MB SPI Flash
400KB SRAM
Up to 15 GPIO pins
Serial interfaces: I2C, SPI, I2S, UART
USB Type C connector
With voltage regulator
Reset and user/firmware buttons
Supply voltage: 5V via USB
Dimensions: 22x18mm approx.

I think that by having a bluetooth module it could be made wireless

SeongGino, 2023-11-04:
Hi, and thanks for the interest! Unfortunately, I can't do a whole lot to help given I don't have one of these boards to test for myself.

What I can say is the hardware at least meets the basic space requirements (though with a concerning low amount of pins), and the ESP32 series [seems to have Arduino IDE compatibility](https://www.electronics-lab.com/getting-started-with-espressifs-esp32-c3-devkitm-1-on-arduino-ide/) so that much checks out. The sketch itself isn't *really* board-dependent on anything except for the USB stack it uses (we only support Arduino native USB and TinyUSB, though most things use the latter anyways), and things related to hardware timers for the cam logic, as all the board info being passed to it is done through [SamcoBoard.h](https://github.com/SeongGino/ir-light-gun-plus/blob/plus/libraries/SamcoPositionEnhanced/SamcoBoard.h); so I would start there with adding entries specific to ESP32 (defining whether it has timers or eeprom storage or not following similar style to the rest of the supported boards) and adding defines to the sketch as needed if it isn't already compatible with functions we use.

As for
> I think that by having a bluetooth module it could be made wireless

While that might sorta *maybe* be possible, Bluetooth mode is not on the roadmap for this project. Given I want to aim for responsiveness and simplicity (like GUN4IR does), BT is the exact opposite - it's slow, would require more processing, can drop out at times without perfect reception (and lightguns require standing away from the screen a fair ways as a rule of thumb), and is not a feature most of the other boards provide out of the box anyways. And unlike force feedback, wireless also isn't nearly a popular enough feature amongst enthusiasts of other systems (most agree that they want quick response times over all else) to be worth pursuing.

Though as with anything else in open source, so long as it's optional, you're free to try your hand at it.

rnrobles, 2023-11-05:
Thanks for the guide, I will try to test using this board, if it is too complicated I would use one of the ones that are already compatible, thanks and greetings

closed 2023-11-05 by rnrobles