SeongGino/ir-light-gun-plus #5: Analog controls

pacomur77, 2023-12-01:
support for analog controls used in guns such as the top shot elite

SeongGino, 2023-12-02:
Right-o! I don't have a gun with analog inputs yet that I can use to test but I'm happy to implement this anyways.

Just one question: How should the stick be recognized?
 - Digital inputs: Would be easiest and can hook into the existing input systems we have.
 - Analog inputs: Would require adding gamepad output support (which LightgunButtons already seems to provide a stub for, interestingly...).

I know GUN4IR had the ability to use a Wii Nunchuk, but I'm not quite sure if they used analog outputs or just converted stick input to digital. The only reason I hadn't considered this earlier was not having a Top Shot myself, which I now realize only uses sticks for directional input.

SeongGino, 2023-12-03:
Then I can absolutely match that behavior (digital), should be fairly simple.

Should a stick be mapped/aliased to keyboard cursor keys, like the Guncon 2's d-pad? Or should it go to a gamepad-like output?

I will probably proto a digital-only implementation before writing in analog output, but admittedly it's hard to think of *gun games* that need an *analog* stick specifically. The obvious example is TC4 retail's FPS mode, but a Guncon 3 isn't even able to be emulated yet in RPCS3.
(I guess maybe the Link's Crossbow Training fans might benefit, aha)
I only want to be sure so I'm not putting too much effort for an extra something that no games practically use.

SeongGino, 2024-01-07:
Top Shot is coming in soon, so I can start implementing this shortly.

Since I've added gamepad output support (mainly to support PCSX2), the idea for now is to make any stick respond as either a d-pad, the keyboard arrow keys, or a second stick - no idea if it should be the right stick or left. Stick-as-cursor is configurable in PCSX2, but apparently games like Railbreak that use gamepad output to support multiplayer with Sindens use a stick in the same way but it isn't clear which axis it's actually checking for that.

Either way, the use of the stick is limited outside of navigation, so it's mostly semantics at this point :P

cmjdbb, 2024-01-12:
The output of the joystick is simulated using the X/Y coordinates from the camera.

SeongGino, 2024-01-12:
> The output of the joystick is simulated using the X/Y coordinates from the camera.

That is indeed what we're doing; I'm just asking to clarify *which stick* should the camera simulate, and so I know which side to map a gun's available analog stick to.

SeongGino, 2024-01-12:
4fcf9d6 has added support for any analog stick to be mapped to the output'd left stick, with the right reserved for the camera when analog output mode is ticked. The pins *should* be set to read at 12-bit precision, but do advise if the sensitivity or the deadzone is too much or not enough (I still have yet to try with an analog stick myself).

Though it seems like we're reaching the limits of what can be safely transmitted with this, as I had to move things around to ensure that camera performance wasn't impacted. For what it's worth, I ran into some strangeness in an early build that caused the board to stall, very much like #3, so please report back if there are any issues.

cmjdbb, 2024-01-13:
I have a suggestion. Both the left joystick and the right joystick should be implemented. In pause mode, use a combination of keys to switch between the left joystick and the right joystick, as some games use the left joystick while others use the right joystick.

cmjdbb, 2024-01-13:
Of course, I have a better suggestion. If it can be switched to the regular relative coordinate mouse mode, it would be unbeatable because there are still many light gun games that only support the regular mouse mode, like "Rambo: The Video Game".

SeongGino, 2024-01-13:
> I have a suggestion. Both the left joystick and the right joystick should be implemented. In pause mode, use a combination of keys to switch between the left joystick and the right joystick, as some games use the left joystick while others use the right joystick.

Hmmm, so different games use different sticks for aiming? That's really dumb--but noted. Will see about implementing soon.
It should be noted that analog output right now is exclusively controlled through a MAMEHOOK-like serial command, `XA`. I'm personally not all that comfortable with making it a shortcut, since it's a workaround I really only implemented for PCSX2 and the problematic few games that don't have multimouse support or an available DemulShooter patch--and at least on Linux, my gun emulation script lets me just send that signal to toggle in and out of cursor-to-stick output mode. :)

As for the mouse, we can only use absolute positioning, since we can only load one type of mouse at a time (and the way mice devices work, they are hard-coded to being either relative or abs at boot)--even then, relative positioning simply doesn't work when lightguns, and the nature of our tracking, are intrinsically based on absolute positioning.

SeongGino, 2024-01-13:
Added stick declaration in 676a69e so the camera & analog stick will swap places when declared. Still exclusively a serial command tho - send `XAL` for camera as left stick, `XAR` for camera as right stick, and just `XA` to toggle it on and off using the last defined output. The default remains camera as right stick/Rx & Ry, and the gun's analog stick as left stick/X & Y.

cmjdbb, 2024-01-14:
How do I switch between joystick and light gun? I have a lot of things to do recently and haven't had time to review the code in detail. You should update the README file to inform everyone how to use the new code.

SeongGino, 2024-01-14:
Just to clarify, *this issue is about adding a physical joystick to the board.* The ability to output the camera as a stick was already implemented but not a part of this issue. I've already tested and that part works as expected.

I'm still waiting for feedback on *this* issue so I can add it to the next release; and because of the scope of what's been added/changed, it will take a while for me to update documentation.

cmjdbb, 2024-01-14:
Sorry，there seems to be a mistake in my description. What I actually meant to ask is how to switch functions between the light gun and the controller. I will find some time to test it out. It seems that there aren't many people who use MAMEHOOK for testing.

SeongGino, 2024-01-14:
Apologies for the confusion.

It should be noted that it (analog output mode) isn't dependent on Mame Hook but I'm not sure how else Windows users can send one-off serial commands. Like I said earlier, it's relatively easy to do on Linux.

*Now that I think about it, I probably should remove the "experimental" label off the Mamehook flag in the sketch code... stability has been fine so long as the patched core is used (and the TinyUSB library is **not** installed in the IDE--I know, confusing). It'll matter less once binaries are being distributed.*

cmjdbb, 2024-01-17:
I tested it and had the following questions:
1: After MAMEHOOK is enabled, the cursor does not move, but the serial port still communicates normally. You need to disconnect the hardware connection and restart to recover.
2: The controller mode does not know how to switch, so it has not been tested.
3: The vibration output signal of MAMEHOOK is wrong. Normally, it should output a high level of 30ms and a low level of 60ms, so that the electromagnet can be released after drawing. Now the output is 60ms high level, 30ms low level, so the electromagnet will hold on.

SeongGino, 2024-01-17:
@cmjdbb None of this is related to the issue at hand for the record, so I have no idea what you're on about right now.

Though I'll only comment on this:
> The vibration output signal of MAMEHOOK is wrong. Normally, it should output a high level of 30ms and a low level of 60ms, so that the electromagnet can be released after drawing. Now the output is 60ms high level, 30ms low level, so the electromagnet will hold on.

That doesn't make any sense because we just take Mamehook's commands and process them at face value?
If you have any other issues that aren't related to the topic here, please make a separate thread for it to declutter *this* one.

SeongGino, 2024-03-15:
Since no one else has opted to comment, I just got around to testing it, and found issues. ef800c41ca65e5d5b73fd20d7022b9979dcdb5c7 resolves sticks being inverted, and da80ceef96e0bf7011a66bf34ef7faefcd1d262d resolves stick position update rate. For Cabela's guns, it should be fine now.

And since [v4.1](https://github.com/SeongGino/ir-light-gun-plus/releases/tag/v4.1) is out, safe to say that release closes this.

closed 2024-03-15 by SeongGino