SeongGino/ir-light-gun-plus #6: IR Leds Diamond arrangement

gustavoalara, 2023-12-02:
Is possible to use the ir leds diamond arrangement like GUN4IR does? I think is more precise than the up-down arrangement

lemmingDev, 2023-12-23:
Planning to set up a TV for this lightgun firmware soon. 
@SeongGino  Is the diamond arrangement a consideration?
Thanks

SeongGino, 2023-12-25:
Camera heuristics and tracking aren't my field of expertise (especially as I'm getting new builds done so I can chip away at the remainder of the TODO over the next couple of weeks), so if anyone wants to take a crack at it I accept pull requests.

Forgot where exactly tracking and the arrangement is handled but I *think* it only needs to be done in the main sketch? Otherwise one could check the DFRobotIRPositionEx library as a second culprit. All the things related to camera is still mostly Prow's original fork verbatim.

cmjdbb, 2024-01-12:
I have compared, the LED arrangement of Samco is more accurate than GUNIR.

lemmingDev, 2024-01-12:
Wow - So the best, and also feature rich lightgun solution is yours!
Open source FTW
I saw a few videos about openfire. Have you guys been in contact?

lemmingDev, 2024-01-12:
I wonder if diamond has any benefit in edge cases like player moving during game play, or faster re-aquisition after reentering screen

lemmingDev, 2024-01-12:
WOW - had not heard of them. Thanks for the tip!
Their 1.0 manual refers to switching between the 2

2024 will be the year of the lightgun!

[image]

SeongGino, 2024-01-13:
Then all the more motivation to have this implemented as a feature! I'd love nothing more than to be able to coexist with other gun systems - and if that means being alongside both Blamcon *and* GUN4IR, all the better.

ackme, 2024-01-23:
Since you're an active member of the Blamcon Discord, you can ask directly instead of being _suspicious_, we've always been happy to answer your questions. From a mathematical perspective 2 up and 2 down are better, I can show you my 3D overlays or you can just browse up in the lightgun-chat channel in the Discord where I outline why and share said overlays. It also means users can buy two Wii bars cheaply if they're building their own, we've always supported the DIY community and tried to provide source your own options.

We only support the diamond configuration because of community demand rather than for a technical superiority reasons, but it is fully implemented.

SeongGino, 2024-01-23:
Hi everyone! Currently busy atm with other life ordeals, but since it all started from this thread, I really should clear some things up.

I've got a tip from Props3D, the developers of the Blamcon system I mentioned earlier, saying they're catching flak of "copying" my system because of me highlighting them in reference to using a "suspiciously similar" format to Samco - which I now realize was extremely bad choice of words and I can only apologize.

That said...

***Guys, gals, & non-binary friends, I'm NOT SAYING Blamcon stole my or Samco code or even copied it.*** If anything, it should be made clear that ***I'm the one that copied Samco in the first place, given the nature of being an open-source fork.***

The Props3D guys are doing great work on their own, much like how me/Prow did our own things for our own work. I only brought up Blamcon to reference *systems and LED arrangements that ours would be compatible with* because I wanted to be able to coexist with other existing lightgun systems. That's where my conversation started and, realistically, where it should've ended.

I just want us to be happy that we have compatible LED arrangements because that can only be good for the lightgun market--which is what this thread about a diamond arrangement was supposed to be about in the first place! Frankly, I'm disappointed that anyone would've taken this in the wrong direction--but because this is a GitHub issues thread and *not* a forum post, that's where I'm going to leave it for now.

...so uh, still seeking PRs for this issue btw. :)

lemmingDev, 2024-01-24:
You and Props3D guys all seem like top blokes
Interesting that some people took your comments that way, or there was any drama, though glad it's sorted

It would be so cool if, like other tech companies do (eg, with USB), all parties got together and set some standards between all guns moving forward

Eg, if everyone could agree to use GX16 connectors with the same pinout and solenoid voltage so that they were ubiquitous, and agreed on a minumum base set of features, that could then be added to to differentiate

Would make it way easier for cab builders, gun makers, firmware writers etc and users in general

SeongGino, 2024-01-27:
> if everyone could agree to use GX16 connectors with the same pinout and solenoid voltage so that they were ubiquitous, and agreed on a minumum base set of features, that could then be added to to [...]

To be fair you're also making the assumption that everyone using GX16 connectors with this example--I don't use any with all three (now soon to be four) of my builds, just a separate USB cable out and a DC pigtail for hotpluggable recoil power input. As far as I'm aware, it all plugs into the same slot/pins on the board anyways. :P 

Things like solenoid voltages also depends on the unit - your Named/Namco arcade gun might take 24V, but the smaller 12V units in most of my builds would overheat much quicker. But that's why arcade builders are there--to work it out amongst themselves.

Still, I suppose the sentiment remains valid. IR LED arrangement is really the biggest pain point I have, since that's going to be the real showstopper preventing guns from physically working alongside one-another on a single machine. I know this project otherwise supports most of the extended features of GUN4IR, and Blamcon I imagine will try to do the same on their end if they haven't already, so that's not necessarily a problem.

But as for the other proprietary systems - specifically Aimtrak & Retro Shooter, they'll just exist in their own universes. And then there's Sinden, who just kinda sorta exists but brings its own software and border requirements which drags down the experience for everyone else involved that doesn't need them.

lemmingDev, 2024-01-29:
Regarding the GX16, I didn't mean to imply that all guns should have them, jut that if they did use them, then a standard pinout should be agreed upon.

I can see someone upgrading from one lightgun system to another, seeing the connector was the same, plugging it in, and 24v being sent to the USB controller instead of 5v

Hopefully a kind soul will submit a pull request to add in diamond arrangement for you one day.

SeongGino, 2024-06-22:
Just cleaning things up here: as most might be able to tell, this project has since moved under the [OpenFIRE](https://github.com/TeamOpenFIRE) banner, and there won't be any more updates on the old GUN4ALL repos.

Appropriately, one of the many updates that project offers over this, is Diamond layout support.

Again, I appreciate everyone's understanding and support.

closed 2024-06-22 by SeongGino