SeongGino/ir-light-gun-plus #7: Could you create a discussion area? Arcade Controls registration is broken

Higgy69, 2023-12-14:
Hi.
Great project. I built mine into a GCon2. I used the existing PCBs to keep existing buttons but routed button wires to RPi Pico. There is not much space left inside!
I have it working once I found my camera 'Top' sticker was in the wrong place!!! :)
I need to sort out software side and IR LED positioning before having a good play.

With RPi Pico not having multicolour LED, would it be possible to say have a simple buzzer to make a sound when in the menu etc? Or can I wire in a separate multicolour LED? Seems RPi Pico has extra GPIOs. 

I wonder if you could create a 'Discussion' area? This was done on the  GitHub MT32-PI project.
Registration is not working on Arcade controls.com (tried multiple emails to register) and I cannot find a contact to tell anyone! Any chance you could drop their Admin a PM to tell them? Sorry I know it is not your problem, I just can't find an easy way to join in this Project discussion without using GitHub 

Thanks again for this project.

SeongGino, 2023-12-14:
Not going to lie, I honestly didn't expect many people outside of the AC boards to see this (given it's the only place I've really publicized this project, unless you happened to come from my Mastodon or something! In which case, uh... (^^;).

But if so, I guess it couldn't hurt!

(And to answer the question: external RGB support isn't implemented yet and is on the TODO list, I just haven't gotten around to doing it yet! But indeed the Pico does have enough analog ports to make that work, even if it might come at the cost of compatibility with upcoming analog stick support.)

SeongGino, 2023-12-14:
Discussions should be open now! The LED support (or lacking thereof atm) can be tracked in #9.

closed 2023-12-14 by SeongGino

SeongGino, 2024-02-12:
@Higgy69 In case you still care about it, I've been notified by the ArcadeControls moderators who wanted me to relay their contact information for you;

scott@k8ch.net / saint@arcadecontrols.com / arcadecontrols@gmail.com
I'm sure they're awaiting a response, given how they went out of their way to notify me recently.

Higgy69, 2024-02-13:
Thanks for that I will send them an email.

Regarding my progress I was having issues on Raspberry Pi (stupid things like arcade image not being able to expand to my full card to therefore add Lightgun games), so I have thought about a small form factor PC for Batat... Whatever that Linux based emulation image is.
Some work ok, but having issues reprogramming keys.
Anyway what I need to do is have a separate setup instead of my daily PC to work on. I have swapped from 3D printed IR modules to modified Wii lightbar with upgraded IR lights.
I know some of the issue is the lightbar positioning, trying to fit in with my daily PC monitor position is not very good due to shelves etc etc clipping the view 

I think my gun element is ok and working. 
Would you suggest RPi or PC/Linux to get the initial setup working?

SeongGino, 2024-02-13:
I don't really use a Pi for much of anything, nor do I use any prefabbed images either, so my bias would lean towards like Linux. But I'm also not the one setting up your system.