SeongGino/ir-light-gun-plus #12: /bin/arm-none-eabi-g++ Error

Fusion-Lightguns, 2023-12-17:
Hello. So i also forked Prow7 and been working on simliar stuff. You have some features I dont so I wanted to try yours out. 

I installed all boards and what not, using a Pi Pico.
I select Pico fix as board and ttry and burn 2.1 hotfixes.

I recieve 
exec: "/bin/arm-none-eabi-g++": file does not exist

Compilation error: exec: "/bin/arm-none-eabi-g++": file does not exist

SeongGino, 2023-12-18:
Are you running Linux? That sounds like missing GCC files - especially since it isn't pointing to a relative Arduino path. It's not something I can reproduce.

Fusion-Lightguns, 2023-12-18:
I am using windows and followed the instruction manual. 

Im running ide 2.1 normal install, imported all libraries and checked i was on right board. 
Not a big deal just wanted to try your code & figured you should know errors.

SeongGino, 2023-12-18:
Then that sounds like an IDE issue. Do other sketches work?
I don't use Windows, so. :woman_shrugging:

SeongGino, 2023-12-20:
Since this seems to be an IDE problem separate from the sketch directly (and no response), I'll be closing this.

closed 2023-12-20 by SeongGino

Fusion-Lightguns, 2024-02-13:
Sorry holidays and life got me caught up for awhile. 

All other sketches worked . I tried my sketches, a blink and different examples. 

I can try
"the meantime, copying the tools folder in rp2040/ to rp2040-tri/ might suffice?"

In the next day or two.

SeongGino, 2024-02-13:
Given that compilation has worked for other users as soon as a few days ago, it's strange that this problem seems to be effecting your environment.

Currently, manual building isn't the recommendation anymore, as prebuilt binaries are distributed now since v3.0, though having build errors is still a problem.

Since editing my last comment, I've updated the core fork's index file so it should be using the `rpipico` directory like upstream--I believe you'll have to remove whatever older version of the core fork you had installed and reinstall the fixed core.

Do let me know if that's fixed it for you.