SeongGino/ir-light-gun-plus #20: Arduino Nano RP2040 Connect

iggut, 2024-02-11:
Hi Seong!
I am having trouble getting this to work with my Arduino nano Rp2040 connect (what I have without spending extra$4). I have 2.1 running and all button inputs are registered (tested with mouse tester webpage). 3.0 for some reason does not work.

My main issue is that I am not sure if the camera is on the correct pins.
I have changed :
   Wire.setSDA(12);
   Wire.setSCL(13);

to match: the board pinout: https://docs.arduino.cc/static/a4d65079ebe3058e37153eb0f93890a8/2f891/pinout.png

Potentially I have the IR leds set up wrong, but looking at the serial console I get the correct message for entering pause mode and then calibration mode. the coordinates do not change when moving the gun, only when pressing A and B buttons (as intended for adjustments)

SeongGino, 2024-02-11:
Hello, and thank you for your report!

Just to clarify: v3.0 added a first-run calibration that's prompting for a pull of the trigger. If you have any leds connected (either Neopixel, though untested, or a 4-pin LED), it should be slowly blinking a color that indicates this. It's very much possible though that you missed this, and my apologies--it might be difficult to communicate that without any visual feedback.

Still, that's an oddity. If the camera pins weren't initiated properly, you'd be seeing an error flooding the serial monitor and wouldn't make it past the init phase.

Since you can access pause mode, is it possible to send an `XT` command through the Serial Monitor and try using the [Samco processing sketch?](https://github.com/samuelballantyne/IR-Light-Gun/tree/master/Samco_4IR_Beta/Samco_4IR_Processing_Sketch_BETA) I'm currently working on a GUI application to interface with v3.0+ which should make setting up a bit smoother, and it's possible that things are a bit wacky with the Nano Connect. If that's so, I'll have to set some overrides for that board specifically.

iggut, 2024-02-11:
I must have the IR leds positioned wrong (also confused on if I needed 2 or 4 sets, made 4 just in case) or maybe because this PC had dual monitors?
If you have the time would you please add two pictures with simple rectangles (one for small monitor, other for bigger TV) and mark the ideal spots to position the LEDs.

SeongGino, 2024-02-11:
Hi, enjoy this five-minute diagram I just made: 
[image]
Purple/Pinkish bits are the emitters; it's basically like two wiimote sensor bars placed at the top/bottom of the display - for TVs, place the emitters a bit further apart from each other to scale if you want more flexible tilt correction.

If indeed you're not getting "Device not responding!" errors in the serial log, then the fault of no-tracking would fall on your emitters. I would recommend plugging in your emitters setup and pointing your phone camera at each one to make sure they are actually emitting light.

Unfortunately, there's no non-dpad-button way to adjust sensitivity on your firmware - I did recently push some fixes for both your Nano board (its builtin LED should now work too) and to be able to send an `XB# - where # = 0/1/2` command over Serial when in pause mode to set IR sensitivity in the current source repo; though obviously you'll have to pull the current source repo and setup your pins in the sketch under line 208 (under the bit that says `#else // For the Raspberry Pi Pico`, I'll prolly make a Nano Connect profile later).

iggut, 2024-02-17:
I must have done something else wrong. Tracking works...well cursor moves. Vertical movements with gun cause horizontal mouse and vice versa. 
Would you be able to help me please. Maybe you have and idea where I messed up?

SeongGino, 2024-02-17:
> Vertical movements with gun cause horizontal mouse and vice versa.

That is a classic case of the camera module being tilted too far to one side! The bane of every first-time DIYer.

It's nothing major, everyone runs into this because the labeling of the camera module is not the most helpful. :)

What's worked for me is tilting the camera so that the "up" label is pointing somewhere around 45-ish degrees counterclockwise. (From the rear/plugs end of the sensor, not the front/lens side) Unfortunately the exact positioning changes from shell-to-shell and sensor module-to-module--you'll have to fiddle with it some, preferably before closing the gun shell shut if possible to make adjustment easier.

Running with the Test/Processing sketch mode on helps.

iggut, 2024-02-17:
You are the best! I lined up that up sticker was exactly visible...lol assumed and you know :P

SeongGino, 2024-02-17:
Fair enough! I just kind of figured that might be the case.

Are you using the current source repo? Any other issues with using the firmware on the Arduino Nano?

iggut, 2024-02-17:
Current source gets not inputs from any buttons for some reason. Switched the same binds code-block to version [IR-GUN4ALL v2.1-hotfix - Fantasyland](https://github.com/SeongGino/ir-light-gun-plus/releases/tag/v2.1-hotfix) and everything works. Calibration took 3 seconds, tracking is perfect and all buttons work.

So not really sure why the current branch does not work.

SeongGino, 2024-02-17:
Gotcha. Mind if you could post/paste the pinout you're using?

iggut, 2024-02-17:
int8_t rumblePin = 17;                            // What's the pin number of the rumble output? Needs to be digital.
int8_t solenoidPin = 16;                          // What's the pin number of the solenoid output? Needs to be digital.
int8_t btnTrigger = 15;                           // Programmer's note: made this just to simplify the trigger pull detection, guh.
int8_t btnGunA = 0;                              // <-- GCon 1-spec
int8_t btnGunB = 1;                              // <-- GCon 1-spec
int8_t btnGunC = 18;                              // Everything below are for GCon 2-spec only 
int8_t btnStart = 19;
int8_t btnSelect = 20;
int8_t btnGunUp = 6;
int8_t btnGunDown = 7;
int8_t btnGunLeft = 8;
int8_t btnGunRight = 9;
int8_t btnPedal = 14;
int8_t btnPump = 13;
int8_t btnHome = 5;

Those and the cam pins:

   Wire.setSDA(12);
   Wire.setSCL(13);

SeongGino, 2024-02-17:
Thanks! And just to be sure, are you using all of these buttons as listed?

I took the liberty of making a test build of v3.0 (so about where current source repo is now) using these mappings with the Arduino Nano set. 
[SamcoEnhanced.arduinoNano-test.uf2.zip](https://github.com/SeongGino/ir-light-gun-plus/files/14320381/SamcoEnhanced.arduinoNano-test.uf2.zip) (Because GitHub doesn't like me shipping UF2 binaries. hmph. :< )

To install UF2 binaries, plug the device into the PC while holding the BOOTSEL button, which should have it show up as a flashdrive; extract the UF2 someplace, then drag and drop the UF2 file onto the drive, where it'll unmount itself once finished and be flashed with the new firmware.

It *should* be compatible with your current calibration table, so just try the buttons and see if they work.

(Just for testing, I only enabled Trigger, A/B/C, and Start/Select, and the force feedbacks--the rest are disabled.)

iggut, 2024-02-17:
Correct for the things enables:  Trigger, A/B/C, and Start/Select, and the force feedbacks

SeongGino, 2024-02-18:
Well damn, I was really banking on that working. :/

Since it works on a Pico, I don't know why the Arduino Nano wouldn't work with the same firmware... And I couldn't tell you why either tbh, since I don't even own a Nano to diagnose the issue.

iggut, 2024-02-18:
Not sure if this will help, but here are the files. 
- [Working](https://github.com/iggut/nixos-config/blob/main/Working%20SamcoEnhanced.ino)
- [Not working](https://github.com/iggut/nixos-config/blob/main/Not%20working%20SamcoEnhanced.ino)

SeongGino, 2024-02-18:
Just the ino doesn't help because there were several library changes between 2.1-x and 3.0, so that's only part of the story - all of the files have to be replaced in their totality.
And again, if the current source works on Pico then it should work on Nano. I'm not getting what's making the Nano be an outlier here...

To really check, you would have to bisect every commit from the last working version (in this case, 2.1-rev whatever) up to where input stops working. Again, I don't have a Nano to personally verify this (though I take donations? Lol) and I'm working on something else now.

iggut, 2024-02-18:
I will bisect every commit and and report back as suggested. Going to have to build a couple of these for friends (not a lie 100% have a friends... :D). Hopefully the info will help others that run into the same.
Probably next Wednesday.

SeongGino, 2024-02-19:
o-*Oh.* I see!
Well then, ah, g'luck I guess!

FWIW you'll probably find some broken commits after cd2949d3e22766110b85adb088194ceb6b5cf399 (i.e. the last commit before the introduction of the extended save system) so you could start from there and then work your way forward/back.

Apologies in advance for the mess--let's just say my habits about verifying code that builds have been, well... *less than ideal* lol.

SeongGino, 2024-02-26:
For what it's worth, v4.0 released with support for both the Arduino Nano with its own official binary, and the [newly released GUI configurator program.](https://github.com/SeongGino/GUN4ALL-GUI)

I bring this up because the GUI has a tab to test button presses, which may help somewhat in diagnosing your issue (though it won't, for example, detect if any pins/buttons are stuck in a pressed state, which could also be a possibility).

Admittedly the layout for the Nano in the GUI may not look very pretty, as again I don't have one to test, but should still work at least to determine what's going on. The non-custom default pins should match the layout you gave me earlier.

SeongGino, 2024-04-24:
It's been close to two months without a response, so I'll just close this for inactivity. Do comment if you still have problems, though.

closed 2024-04-24 by SeongGino